BasicAirData/AirDataComputer #31: Blend the ADC and GPSLogger data

JLJu, 2016-11-29:
It should be interesting to have a standardized workflow to be able to use the data from the ADC together with the data from the GPSLogger ([check here](http://www.basicairdata.eu/2016/12/17/import-gpslogger-tracks-within-qgis/), link live after 17/12/16)). Ideally, all the data should converge at the same software for visualization purposes. Preliminary tests showed that a GIS can handle our data.

GrazianoCapelli, 2017-04-13:
I'm in. With great pleasure!

JLJu, 2017-04-21:
As soon as the communication library is ready we will be able to discuss this issue. At a glance, it consists in adding ADC data to GSPLogger DB and add a new export format.

JLJu, 2017-04-27:
Evolved in #45

closed 2017-04-27 by JLJu

GrazianoCapelli, 2017-04-27:
For sure it's better to create and mantain a new dedicated app instead of add features to the GPSLogger.
We could re-use many pieces of code of GPS Logger anyway.

I really like the Unix philosophy :grinning: