BasicAirData/AirDataComputer #42: Revision of ADC Communication protocol

JLJu, 2017-03-16:
Under the point of view of bandwidth a "Monolithic" approach to data exchange can be a problem over a wireless link, if required sample rate is high.
Check message #10 within Rev 1 
[ADC DOC](https://github.com/BasicAirData/AirDataComputer/blob/master/Software/Documentation/Rev%201%20ADC%20Common%20Message%20Set.odt)
I think is better instead to give the possibility to the user to receive subsets of data. In such a way it will be possible to arrange for limited bandwidth situations, low power requirements, and fast sample rate.

GrazianoCapelli, 2017-03-16:
A couple of notes/proposal about the $DTA message:

1) In case, could we consider to move the timestamp in the first fields? Timestamp is fundamental and present in any set of data, and it is the first check to do in order to find duplicates and, in case, reorder data. Useful in case of communication interruption;

2) The message could remain the same also in case you want to receive a subset of data: we could simply send the message sending commas without put any values between them. Could be a good start to limit bandwidth without introduce any complication. The $DTQ request could contain a series of 1 and 0, depending on data fields you requires.

JLJu, 2017-03-16:
1. Ok. 
2. Yes, the simpler the better. Vector mask ok :-)

JLJu, 2017-03-16:
File updated [ADC DOC](https://github.com/BasicAirData/AirDataComputer/blob/master/Software/Documentation/Work%20in%20progress%20ADC%20Common%20Message%20Set.odtt)

closed 2017-04-28 by JLJu