BasicAirData/AirDataComputer #45: Make a simple android app to test ADC communication via Bluetooth

GrazianoCapelli, 2017-04-26:
I propose to make a first android app (a simple **ADC Tester**), in order to:
- Verify and test the ADC communication protocol;
- Find possible bottlenecks and limits of the Bluetooth interface;
- Refine protocol, transmission methods, and connection setup in order to optimize the system;
- Find the best way to implement the communication on Android-side;
- (Last but not least) Give to all users a fast and simple way to test their own DIY units;

JLJu, 2017-04-26:
That will be great! Not to say that this will be the first fully-fledged plug and play DIY solution :-). Of course, consider me on board. I can help with Android Apps. 
The basic test of libraries is proceeding and will end soon. Update of documentation is next to completion.

GrazianoCapelli, 2017-05-05:
A small app has been committed (commit a30a686) in order to offer a fast way to test the device via Bluetooth. The app can connect / disconnect, send and receive messages, and presents the messages in a Terminal-like way.
This is obviously a first start point, it needs some (many) adjustments and refinements.

JLJu, 2017-05-09:
A screen shot of ADCTester communicating with the ADC. By far the best! :-)
[image]

closed 2017-07-23 by JLJu