BasicAirData/AirDataComputer #57: Review the File dump method

GrazianoCapelli, 2018-01-10:
The Dump method (from SD to the serial interfaces) now have some problems:
- It fails on large files because it fills the buffer and discards the data;
- It is a blocking operation that needs the possibility to abort it;

JLJu, 2018-01-10:
Current file download is bulky but easy to implement Teensy side and Android/BT/PC side. Unfortunately, to block the transfer you should cut the power to the ADC. 

Move to a standard protocol for file transfer would allow us to use standard serial terminal software, that can be comfortable and also allow automatic scripts to be executed over the serial connection.
Zmodem implementation seems to be available for Teensy
https://forum.pjrc.com/threads/26646-Zmodem-transfer-from-Teensy-to-PC
https://forum.arduino.cc/index.php?topic=382044.0

GrazianoCapelli, 2018-01-10:
For now I implemented an easy way to abort the dump (commit 7319735).

Now it is possible to end the dump anytime, by sending another command from the same serial interface.
The dump will send the `$EOF` sentence and will terminate.
The new command sent will be processed as usual.

It is possible to know if the file is fully downloaded, comparing the remote and the local size.
I take a look to your link and, basing on that code, and I choose to add a periodic `flush()` during the dump process.

The dump method now seems to be rough but fully functional.
To be deeply tested!

JLJu, 2018-01-11:
Way to go!

closed 2018-01-12 by JLJu