BasicAirData/AirDataComputer #97: Help with 3D model rendering and publishing

JLJu, 2019-02-01:
[During the brainstorming phase](https://www.basicairdata.eu/projects/air-data-systems/brainstorming-air-data-system-winter-2018-2019/), we often need to play around with 3D models and render. Blender is welcome.
The idea is to provide some graphical follow up.
The final models will be published during the release phase.

Lucasnribeiro, 2019-03-03:
Hi! I have some experience with blender. I could help with modeling + rendering. Though, I never modeled for engineering, wich requires real life accurate measures. Could you give more detail on what you expect to be produced? Thanks!

JLJu, 2019-03-03:
Hi Lucas!
Thank you for your interesting in Basic Air Data.

We miss, but not only, the 3D drawing files of the probes used with the [ADC](https://www.basicairdata.eu/projects/skyeye/review-diy-air-data-computer-asgard-and-pitot-static-on-skyeye-rc-model/) in a neutral format.

1. - [Pitot Probe](http://www.basicairdata.eu/knowledge-center/design/pitot-static-probe/)
   [Reference drawings, page 31/32](http://www.basicairdata.eu/attachments/Manuals/8mmESP-PITOT-Manual.pdf)
   [Reference Github repository](https://github.com/BasicAirData/8mmPitot-StaticProbe)
2. - [Air Data Boom](https://www.basicairdata.eu/projects/air-data-boom/)
[Reference drawings](https://github.com/BasicAirData/AirDataBoom/tree/master/PDFBuildPlans)

For each probe, we need, in this order

1. Each component 3D model
Editable source file
STEP file
STL file

2. Assembly model
Editable source file
STEP file
STL file

3. Animation that illustrates the working principle

Unit of measurement for the drawings models should be millimeters. 

For the sake of simplicity, it is better to split the contributions in different pull requests, for example, one component at a time.

Let me know what probe you are willing to contribute to, and I will provide further details.

Lucasnribeiro, 2019-03-04:
Seems that the Pilot Probe could be a nice start. I'm looking at the reference drawing to make the model in Blender with accurate measures.

JLJu, 2019-03-04:
[Have also a look at the pictures of the Pitot](http://www.basicairdata.eu/knowledge-center/design/pitot-static-probe/). The protruding pipe and head are always the same, no matter if a fixation flange is used.

JLJu, 2019-03-04:
P.S. Note that there is some material already posted [in this repo](https://github.com/BasicAirData/8mmPitot-StaticProbe/tree/master/Drawings)

Lucasnribeiro, 2019-03-05:
[image]

So, I struggled a bit to understand how engineering drawing works. Did some mistakes with the diameters and took longer than I expected. But at the moment im almost certain that it is accurate. 

I was modeling based only on the drawings, but now that i've downloaded the STL files that you already had it made things more clear. Like the holes ( that I need to make ) on the main tube. They go around it all the way.

But comparing the amount of faces on the STL model and the one that i've made we can see an improvement. The only thing that is concerning me right now is that Blender can't output STEP files. Only STL. Reading online I've found that CAD programs can convert STL to STEP, but since I never used CAD software before I imagine that you could give some help with it.

Here's the model so far: https://ufile.io/xwntq

I still have to make some adjustments, but all the measures are as the drawings. Scale is in mm.

JLJu, 2019-03-06:
Nice!
[image]
Is it possible to have separate surfaces for the different parts? I mean, the pitot head part seems flat at bottom.

A little background. The models you are working on are intended to be used for virtual assembly, 3D print,  web illustrations, videos of working principle ([check this out, not a pitot :-)](https://youtu.be/oUd4WxjoHKY?t=81)), and general render with the ADC.

Freecad(that is free) is able to convert STL to STEP, it's not a concern right now.

Lucasnribeiro, 2019-03-06:
Oh Im sorry for that. The flat bottom is because I have different layers on the Blender project and STL export probably doesn't support saving just one of them and saves it all merged. But the correct model has the holes.

The pitot is a separate part, so as the body and the inner tube. They can all be separated, cut and edited. When finished I will create files for each of them.

JLJu, 2019-03-06:
It's okay. I think it is better to focus on the model for the pitot without flange (page 31). This probe has a visible epoxy sealing at one end of the main 8 mm pipe.
Regards